bcnjr5/linux-pitop-battery #3: Warning during install

rricharz, 2017-03-08:
See he following warning during installation in standard Raspbian Jessie:
`Loading new i2c-gpio-param-1.0 DKMS files...
First Installation: checking all kernels...
dpkg: warning: version '*-*' has bad syntax: version number does not start with digit
It is likely that 4.4.50-v7+ belongs to a chroot's host
Module build for the currently running kernel was skipped since the
kernel source for this kernel does not seem to be installed.
`

rricharz, 2017-03-08:
Unfortunately the installation is broken on my copy of Raspbian Jessie, even after
sudo apt-get update
sudo apt-get upgrade
rpi-update

rricharz, 2017-03-09:
The problem is loads the wrong kernel headers. A work-around is installing the proper kernel source BEFORE executing dpkg. One gets still lots of warnings, and the wrong headers are loaded, but the installation completes successfully.

Output:
```
sudo apt-get install pitop-battery-support
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  cpp-4.6 dkms gcc-4.6 i2c-gpio-param-dkms linux-headers-3.6-trunk-common
  linux-headers-3.6-trunk-rpi linux-kbuild-3.6 sbs-battery-dkms
Suggested packages:
  gcc-4.6-locales libmudflap0-4.6-dev gcc-4.6-doc libgcc1-dbg libgomp1-dbg
  libquadmath-dbg libmudflap0-dbg libcloog-ppl1 libcloog-ppl0 libppl-c4
  libppl12 libppl9
Recommended packages:
  linux-headers-686-pae linux-headers-amd64 linux-headers-generic
  linux-headers pt-hub-controller
The following NEW packages will be installed:
  cpp-4.6 dkms gcc-4.6 i2c-gpio-param-dkms linux-headers-3.6-trunk-common
  linux-headers-3.6-trunk-rpi linux-kbuild-3.6 pitop-battery-support
  sbs-battery-dkms
0 upgraded, 9 newly installed, 0 to remove and 1 not upgraded.
Need to get 13.3 MB of archives.
After this operation, 49.8 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
WARNING: The following packages cannot be authenticated!
  i2c-gpio-param-dkms sbs-battery-dkms pitop-battery-support
Install these packages without verification? [y/N] y
Get:1 http://0xc9.net/apt-deb/ ./ i2c-gpio-param-dkms 1.0 [5,300 B]
Get:2 http://mirrordirector.raspbian.org/raspbian/ jessie/main cpp-4.6 armhf 4.6.4-5+rpi1 [4,145 kB]
Get:3 http://0xc9.net/apt-deb/ ./ sbs-battery-dkms 1.0 [8,254 B]          
Get:4 http://0xc9.net/apt-deb/ ./ pitop-battery-support 1.0 [5,458 B]          
Get:5 http://mirrordirector.raspbian.org/raspbian/ jessie/main dkms all 2.2.0.3-2 [70.9 kB]
Get:6 http://mirrordirector.raspbian.org/raspbian/ jessie/main gcc-4.6 armhf 4.6.4-5+rpi1 [4,363 kB]
Get:7 http://mirrordirector.raspbian.org/raspbian/ jessie/main linux-headers-3.6-trunk-common armhf 3.6.9-1~experimental.1+rpi7 [4,189 kB]
Get:8 http://mirrordirector.raspbian.org/raspbian/ jessie/main linux-kbuild-3.6 armhf 3.6-1~experimental.1+rpi2 [135 kB]
Get:9 http://mirrordirector.raspbian.org/raspbian/ jessie/main linux-headers-3.6-trunk-rpi armhf 3.6.9-1~experimental.1+rpi7 [414 kB]
Fetched 13.3 MB in 16s (827 kB/s)                                              
Selecting previously unselected package cpp-4.6.
(Reading database ... 121685 files and directories currently installed.)
Preparing to unpack .../cpp-4.6_4.6.4-5+rpi1_armhf.deb ...
Unpacking cpp-4.6 (4.6.4-5+rpi1) ...
Selecting previously unselected package dkms.
Preparing to unpack .../dkms_2.2.0.3-2_all.deb ...
Unpacking dkms (2.2.0.3-2) ...
Selecting previously unselected package gcc-4.6.
Preparing to unpack .../gcc-4.6_4.6.4-5+rpi1_armhf.deb ...
Unpacking gcc-4.6 (4.6.4-5+rpi1) ...
Selecting previously unselected package i2c-gpio-param-dkms.
Preparing to unpack .../i2c-gpio-param-dkms_1.0_all.deb ...
Unpacking i2c-gpio-param-dkms (1.0) ...
Selecting previously unselected package linux-headers-3.6-trunk-common.
Preparing to unpack .../linux-headers-3.6-trunk-common_3.6.9-1~experimental.1+rpi7_armhf.deb ...
Unpacking linux-headers-3.6-trunk-common (3.6.9-1~experimental.1+rpi7) ...
Selecting previously unselected package linux-kbuild-3.6.
Preparing to unpack .../linux-kbuild-3.6_3.6-1~experimental.1+rpi2_armhf.deb ...
Unpacking linux-kbuild-3.6 (3.6-1~experimental.1+rpi2) ...
Selecting previously unselected package linux-headers-3.6-trunk-rpi.
Preparing to unpack .../linux-headers-3.6-trunk-rpi_3.6.9-1~experimental.1+rpi7_armhf.deb ...
Unpacking linux-headers-3.6-trunk-rpi (3.6.9-1~experimental.1+rpi7) ...
Selecting previously unselected package sbs-battery-dkms.
Preparing to unpack .../sbs-battery-dkms_1.0_all.deb ...
Unpacking sbs-battery-dkms (1.0) ...
Selecting previously unselected package pitop-battery-support.
Preparing to unpack .../pitop-battery-support_1.0_all.deb ...
Unpacking pitop-battery-support (1.0) ...
Processing triggers for man-db (2.7.0.2-5) ...
Processing triggers for systemd (215-17+deb8u6) ...
Setting up cpp-4.6 (4.6.4-5+rpi1) ...
Setting up dkms (2.2.0.3-2) ...
Setting up gcc-4.6 (4.6.4-5+rpi1) ...
Setting up i2c-gpio-param-dkms (1.0) ...
Loading new i2c-gpio-param-1.0 DKMS files...
First Installation: checking all kernels...
dpkg: warning: version '*-*' has bad syntax: version number does not start with digit
It is likely that 4.9.13-v7+ belongs to a chroot's host
Building initial module for 4.9.13-v7+
Done.

i2c-gpio-param:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.9.13-v7+/extra/

depmod.....

DKMS: install completed.
Setting up linux-headers-3.6-trunk-common (3.6.9-1~experimental.1+rpi7) ...
Setting up linux-kbuild-3.6 (3.6-1~experimental.1+rpi2) ...
Setting up linux-headers-3.6-trunk-rpi (3.6.9-1~experimental.1+rpi7) ...
Examining /etc/kernel/header_postinst.d.
run-parts: executing /etc/kernel/header_postinst.d/dkms 3.6-trunk-rpi
Setting up sbs-battery-dkms (1.0) ...
Loading new sbs-battery-1.0 DKMS files...
First Installation: checking all kernels...
dpkg: warning: version '*-*' has bad syntax: version number does not start with digit
It is likely that 4.9.13-v7+ belongs to a chroot's host
Building initial module for 4.9.13-v7+
Done.

sbs-battery:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.9.13-v7+/kernel/drivers/power/supply/

depmod....

DKMS: install completed.
Setting up pitop-battery-support (1.0) ...
Processing triggers for systemd (215-17+deb8u6) ...
```

bcnjr5, 2017-03-09:
Sorry, I'm not sure where the `dpkg: warning: version '*-*' has bad syntax: version number does not start with digit` comes from, but it happens on my build too, and doesn't seem to cause any problems.

I'll prepend `apt-get install raspberrypi-kernel-headers` to the install instructions.

rricharz, 2017-03-09:
I had to install this https://github.com/notro/rpi-source to make it work (the whole source unfortunately). `apt-get install raspberrypi-kernel-headers` did not install the right ones. It is really annoying that this does not work right on Raspbian.

bcnjr5, 2017-03-09:
Hmm... Do you have the `raspberrypi-kernel` package installed? (It seems to be the default)

rricharz, 2017-03-09:
Yes, see
sudo apt-cache policy raspberrypi-kernel
raspberrypi-kernel:
  Installed: 1.20170303-1
  Candidate: 1.20170303-1
  Version table:
 *** 1.20170303-1 0
        500 http://archive.raspberrypi.org/debian/ jessie/main armhf Packages
        100 /var/lib/dpkg/status

bcnjr5, 2017-03-09:
What does `uname -r` return?

On my device, the `rpi-update` utility mentioned moving to the `4.9` tree (and your error says that it is trying to build for `4.9.13-v7+`), but the actual running version (for Raspbian) is `4.4.50-v7+`.

bcnjr5, 2017-03-09:
Also, what model are you using? (Mine is a Raspberry Pi 3 Model B)

bcnjr5, 2017-03-09:
_Also_, since you said you got it working on your end, what is the output of `dkms status`?

Mine is
```
i2c-gpio-param, 1.0, 4.4.50-v7+, armv7l: installed
sbs-battery, 1.0, 4.4.50-v7+, armv7l: installed
```

rricharz, 2017-03-10:
`rpi-update` moving to the unreleased 4.9 tree was the problem. I went back to 4.4.50-v7 and now both installation with 'apt-get' and manual work fine. Executing 'sudo apt-get install raspberrypi-kernel-headers' BEVORE doing the actual installation is mandatory in both cases!

commit 5e8771f0 (2017-03-09): Added missing package

Fixes #3
closed 2017-03-10 by rricharz