bcnjr5/linux-pitop-battery #6: Shutdown if battery low

rricharz, 2017-03-10:
The rpi needs to be shut down gracefully if the battery level gets low. The lxde lxpanel battery module just gives a warning, but does not shut down. Perhaps a script like unix.stackexchange.com/questions/84437 could do the job. The info is at sys/class/power_supply/sbs-1-000b

bcnjr5, 2017-03-10:
On Debian-based systems, a seperate utility such as [UPower](http://unix.stackexchange.com/a/317933) should be used.

Here is an example configuration file:

```sh
# Only the system vendor should modify this file, ordinary users
# should not have to change anything.

[UPower]

# Enable the Watts Up Pro device.
#
# The Watts Up Pro contains a generic FTDI USB device without a specific
# vendor and product ID. When we probe for WUP devices, we can cause
# the user to get a perplexing "Device or resource busy" error when
# attempting to use their non-WUP device.
#
# The generic FTDI device is known to also be used on:
#
# - Sparkfun FT232 breakout board
# - Parallax Propeller
#
# default=false
EnableWattsUpPro=false

# Don't poll the kernel for battery level changes.
#
# Some hardware will send us battery level changes through
# events, rather than us having to poll for it. This option
# allows disabling polling for hardware that sends out events.
#
# default=false
NoPollBatteries=false

# Do we ignore the lid state
#
# Some laptops are broken. The lid state is either inverted, or stuck
# on or off. We can't do much to fix these problems, but this is a way
# for users to make the laptop panel vanish and for programs like
# gnome-settings-daemon to not suspend on system startup.
#
# default=false
IgnoreLid=true
# I don't think the Pi-Top has a lid sensor.

# Policy for warnings and action based on battery levels
#
# Whether battery percentage based policy should be used. The default
# is to use the time left, change to true to use the percentage, which
# should work around broken firmwares.
# default=false
UsePercentageForPolicy=true

# When UsePercentageForPolicy is true, the levels at which UPower will
# consider the battery low, critical, or take action for the critical
# battery level.
#
# This will also be used for batteries which don't have time information
# such as that of peripherals.
#
# If any value is invalid, or not in descending order, the defaults
# will be used.
#
# Defaults:
# PercentageLow=10
# PercentageCritical=3
# PercentageAction=2
PercentageLow=10
PercentageCritical=5
PercentageAction=3

# When UsePercentageForPolicy is false, the time remaining at which UPower
# will consider the battery low, critical, or take action for the critical
# battery level.
#
# If any value is invalid, or not in descending order, the defaults
# will be used.
#
# Defaults:
# TimeLow=1200
# TimeCritical=300
# TimeAction=120
TimeLow=1200
TimeCritical=300
TimeAction=120

# The action to take when "TimeAction" or "PercentageAction" above has been
# reached for the batteries (UPS or laptop batteries) supplying the computer
#
# Possible values are:
# PowerOff
# Hibernate
# HybridSleep
#
# If HybridSleep isn't available, Hibernate will be used
# If Hibernate isn't available, PowerOff will be used
CriticalPowerAction=PowerOff
```

rricharz, 2017-03-11:
I agree. UPower works and shuts down nicely, except that it does not show the messages on the lxde desktop. It is certainly much better than having no power management. I have opened a branch and have put the info there.

rricharz, 2017-03-11:
Any idea on how to proceed on this issue, or do you want to merge the branch and close the issue? For me it is good enough for now. The main purpose, to protect the battery and to shut down gracefully before the battery pack cuts the power independent on whether one works in a gui or not has been achieved.

bcnjr5, 2017-03-11:
I'll merge it in.
I think it might be trivial to add a notification using the freedesktop.org intrface, so I'll leave the issue open for the moment.

bcnjr5, 2017-03-11:
Would you prefer to track the percentage instead of the time remaining?

As an example, in the configuration file I posted above, I used the following values:

```
UsePercentageForPolicy=true
PercentageLow=10
PercentageCritical=5
PercentageAction=3
```

I did that because the clock on my display always seems to say `00:00` when discharging.
(It works fine when charging.)

rricharz, 2017-03-12:
Percentage is fine. The problem is that the bq40z60 cannot calculate a time to empty during charging because it cannot measure the discharging current.

I would prefer PercentageAction=5 to be definitely on the safe side. I have seen a battery cutting the plug near 3% because one cell went too low already, and gauching was not yet perfect.

commit 7dc10747 (2017-03-12): Added percentage instructions

Closes #6
closed 2017-03-12 by bcnjr5