bcnjr5/linux-pitop-battery #7: not working

clehner, 2017-03-31:
I followed the "Manual install method", updated config.txt and rebooted, but the battery device does not show up. 

I'm running Debian jessie with some packages from testing.
```
$ uname -r
4.4.50-v7+
$ ls /lib/modules/$(uname -r)/build
arch   certs   Documentation  firmware  include  ipc      kernel  Makefile  Module.symvers  samples  security  tools  virt
block  crypto  drivers        fs        init     Kconfig  lib     mm        net             scripts  sound     usr
```
```
$ lsmod|grep 'i2c\|sbs\|gpio'
sbs_battery             7010  0
i2c_algo_bit            5495  0
i2c_dev                 5859  0
i2c_gpio_param          2815  0
bcm2835_gpiomem         2976  0
```

```
$ grep ^dtparam /boot/config.txt
dtparam=spi
dtparam=audio
```

dmesg:
```
...
[  230.805650] i2c /dev entries driver
[  230.835898] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[  230.835930] brcmfmac: brcmf_add_if: ignore IF event
[  230.853835] i2c-gpio i2c-gpio.1: using pins 2 (SDA) and 3 (SCL)
[  232.785613] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  232.858125] cfg80211: Regulatory domain changed to country: US
[  232.858150] cfg80211:  DFS Master region: FCC
[  232.858160] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[  232.858175] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 3000 mBm), (N/A)
[  232.858190] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2300 mBm), (N/A)
[  232.858205] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2300 mBm), (0 s)
[  232.858218] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2300 mBm), (0 s)
[  232.858231] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 3000 mBm), (N/A)
[  232.858243] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[  232.873282] thermal thermal_zone1: failed to read out thermal zone (-19)
[  232.873350] sbs-battery 1-000b: sbs-battery: battery gas gauge device registered
[  232.873397] i2c i2c-1: new_device: Instantiated device sbs-battery at 0x0b
[  232.958035] i2c i2c-1: sendbytes: NAK bailout.
[  233.090850] i2c i2c-1: sendbytes: NAK bailout.
```
Any suggestions?

bcnjr5, 2017-04-01:
Based on these lines from dmesg,

```
[  232.873350] sbs-battery 1-000b: sbs-battery: battery gas gauge device registered
[  232.873397] i2c i2c-1: new_device: Instantiated device sbs-battery at 0x0b
```

it seems that your battery is detected and registered.

Have you set up some way to actually _read_ the battery?

All this repository does is set up `/sys/class/power_supply/sbs-1-000b` so that other Linux applications (such as battery widgets) are aware that there is a battery connected to the system. Most desktop environments automatically detect power supplies and add an icon to a taskbar when one is connected.

What desktop environment are you using? (Gnome, KDE, LXDE, window manager + some programs, etc.)

clehner, 2017-04-04:
@bcnjr5 cool, i see that in /sys now. i guess i wasn't looking in the right place before. i'm using dwm+[slstatus](https://github.com/drkhsh/slstatus) - i updated the battery status config for `sbs-1-000b` and am now able to read the battery status successfully.

i had made another thing which shims the power_supply sys device files: [pibatd](https://git.scuttlebot.io/%259ySUSgdxBOj67eNwjfJt0fmnTOBYviiB1iouJ3PRZy8%3D.sha256/blob/90ec9115a6fff4bf94919ebd258b1ab388f212f4/pibatd.c). it relies on polling though which isn't great, so i'm happy to deprecate it for a proper driver. thanks for making this thing.

closed 2017-04-04 by clehner

bcnjr5, 2017-04-04:
Glad to be of assistance!