bcnjr5/linux-pitop-battery #9: app not working after battery firmware upgrade

mvdm1, 2017-08-05:
First of all best battery indicator for the pi-top, many thanks!

After doing a firmware upgrade / diagnostics on a battery that wouldn't charge, I can't seem to get this app to work again.  I now seem to be unable to disable the I2C drivers.  The commented i2c-dev and dtparam=i2c_arm=on seem to automatically uncomment them selves.  I also tried deleting them from the files, but they append to the end of the file.  Not sure if this is related to trying to install and run the pi-top-batery-analyse and its prerequisites.

[https://github.com/rricharz/pi-top-battery-status/blob/master/battery_problems.md](https://github.com/rricharz/pi-top-battery-status/blob/master/battery_problems.md)

[https://github.com/rricharz/pi-top-battery-analyze](https://github.com/rricharz/pi-top-battery-analyze)

FYI... to fix a battery that won't charge, make a script that continuously runs the firmware update script, run it, and leave it over night.  Eventually my battery stopped failing the firmware update with error 120.

rricharz, 2017-08-05:
You can try to uninstall the pigpio library, which is used by pi-top-battery-analyze, using:

sudo apt-get purge pigpio

Afterwards you should be able to comment out the standard i2c drivers.

mvdm1, 2017-08-05:
Hi Rricharz,

Thanks for the quick response, I just uninstalled pigpio, rebooted and I still have the same result.  Do you have any other suggestions I can try?

Thanks in advance

rricharz, 2017-08-06:
Linux pi-top-battery allows to use the standard battery monitor at the expense of being incompatible with most raspberry pi software using the i2c bus. You must have some other such software running. It is certainly not my analyze program. Have you installed the pi-top speaker software from pi-top? The newest version uses a deamon which searches for pi-top hardware and might turn on i2c to search for other pi-top hatdware. I propose that you burn a new image, install software step by step to find out what interferes.

mvdm1, 2017-08-06:
Hi rricharz,

Thanks for the help, I really appreciate it.  I completely understand the situation but it really sucks that i have to create the image again.  

I know this is unrelated to the pitop-battery but I'm wondering if there is a way to deny whatever script that is changing the config file.  I already did:
    sudo chattr +i /etc/modprobe.d/raspi-blacklist.conf 
    sudo chattr +i /etc/modules

The above works, and whatever script is not able to change them, but  with sudo chattr +i /boot/config.txt i get the following:
chattr: Inappropriate ioctl for device while reading flags on /boot/config.txt

I suspect that this may fix my problem if I can make the /boot/config immutable, do you maybe have any suggestions of how I can do this? 

Thanks in advance

mvdm1, 2017-08-06:
Found the culprit through grep " install " /var/log/dpkg.log:

sudo apt-get purge pt-peripheral-cfg

mvdm1, 2017-08-06:
After I found the culprit I just had to uninstall pitop-battery-support
sudo apt-get autoremove pitop-battery-support
openbox --reconfigure
sudo reboot

Follow your installation instructions again, and reboot and voila!

closed 2017-08-06 by mvdm1